GIPdA/ZetaRF #1: Qn: Wiring diagram for this library...

adamfowleruk, 2017-03-16:
Hello,

Thanks for writing this library - I wasn't looking forward to hand cranking it myself. It doesn't seem to work for me though. I assume I've not wired my Arduino Pro Mini to the Zeta chip correctly, or it's because I've got the 433 MHz chip.

Could you please add to your repository anything that may help me. E.g. photo of a correctly wired up module to an Arduino, wiring diagram, or guidance around 433 MHz settings.

I'm working on a data radio project for a youth group I work with.

Thanks very much in advance.

GIPdA, 2017-03-16:
Hello,

You're welcome, it's a quite nice little rf module :-)

The problem you have is because the library, by default, is configured for 868MHz (config header). I will have to add an option to change that...

To fix that it should be as easy as changing the central frequency using WDS (Wireless Development Suite) and the config xml I provided (si4455_revb1_bidir_fixed_crc_pre10_sync4_pay8.xml), to generate the right config header for the 433MHz module.

I just generated one (had to zip it to upload it here), please try it (replace the existing one) and let me know if it works :
[radio_config_fixed_crc_pre10_sync4_pay8.h.zip](https://github.com/GIPdA/ZetaRF/files/848410/radio_config_fixed_crc_pre10_sync4_pay8.h.zip)

The 433MHz module should have the same hardware pretty much I think (same XTAL), so it should work. Otherwise I will look into that a bit more later.

adamfowleruk, 2017-03-16:
Hi,

Thanks for the quick reply. I'd just generated a 433 MHz header based on your existing one myself. I also introduced a new FREQ_433 define to switch which WDS include I use. I still have problems though.

The waitCTS() function appears to never exit - it doesn't seem to receive the interrupt from the Zeta chip. This is why I think I've got some wiring issues.

I've uploaded a diagram of my wiring (Twitter was just the easiest way to do it...) https://twitter.com/adamfowleruk/status/842445225469579264

If you could please take a quick look - I imagine it's a really simple fix.

Once I have it working I'll test both the variable and fixed packet configurations on the 433 MHz and send you a pull request for these two new header files, the #define for FREQ_433, and an updated README.

Thanks very much for your time.

GIPdA, 2017-03-16:
I see, I'm pretty sure you inverted MISO and MOSI on your wiring. SDO should be connected to MISO (Master In) and SDI to MOSI (Master Out).

Be aware that the variable packet config is not very reliable at the moment, I had troubles with it and didn't had time to do more testings. It might be due to the packet config (preamble, etc) which are optimized for a very low latency (a few milliseconds) and not reliability... I might provide other configs one day.
But I'm interested if you can test it, maybe you'll have other results.

I also made some changes that I need to push, like a timeout in the waitCTS(). Will do that soon.

adamfowleruk, 2017-03-16:
Hello,

Thanks for the advice. I had connected the Arduino as you said, but given the Arduino is acting as the master and not the slave I swapped the MISO and MOSI wiring around - and it worked! 

I also commented out my changes (using delay instead of CTS) to sendMessage, and can confirm that this works on the 433 MHz version... at least when you wire up the project correctly!

I've not tested the variable length piece though.

I'll write up a quick blog entry, then I'll generate the relevant 433MHz configurations, add the FREQ define, and send you a pull request.

As I build my project I'll continue testing and sending you any modifications I need to make.

Thanks again for all your help, and providing this excellent library!

GIPdA, 2017-03-16:
Always better when the wirings are correct :-p Glad it works!

Good to know the 433MHz module is supported as well, I may use one in the future. Somebody to test the 915MHz one? :-D

I just pushed some modifications, CTS timeout included, may be useful to you.
I'll merge your push request to add support to the 433MHz module then.

Good luck with your project, let me know if you run into any trouble with the lib or if something is missing!

closed 2017-03-19 by GIPdA