GIPdA/ZetaRF #3: Problems with 434Mhz

regisaabdalla, 2018-03-22:
Hi!
First, thank you for share!
you did a wonderfull work! 

Can you help me with a issue?
For some reason the project don´t work with 434Mhz!!
I have a RF stick (EZRadio 4455Cled-434) in 434 Mhz and like to test it with your Project and my zeta (ZETA-1 ver B).

I read this info from ZETA:
Read IT status:
Chip rev: 17 ,      Part    : 21828    ,  PBuild  : 0 ,      ID      : 3328, Customer: 0,       Rom ID  : 3    ,      Bond    : 0, Rev Ext   : 3 ,       Rev Branch: 0 ,     Rev Int   : 13,   Patch     : 0, Func      : 1 ,        SVN Flags : 0  ,     SVN Rev   : 3540385792

I believe it´s about config file.... can you help?
I attached the config file to this post.

Thanks in advance. 
Besta regards,
Regis

[radio_config_fixed_434.zip](https://github.com/GIPdA/ZetaRF/files/1835759/radio_config_fixed_434.zip)

beinnlora, 2018-03-22:
Try first hardcoding the required config file instead of relying on compile-time definitions,

edit ZetaRF.cpp and replace  lines 15-30 fom

```
#ifdef VARIABLE_LENGTH_ON
#ifdef ZETARF_FREQUENCY_433MHZ
    #include "configs/radio_config_vl_crc_pre10_sync4_pay8_433mhz.h"
#else
    #include "configs/radio_config_vl_crc_pre10_sync4_pay8.h"
#endif
    #warning Using variable length packets
#else
// Fixed size packet
#ifdef ZETARF_FREQUENCY_433MHZ
    #include "configs/radio_config_fixed_crc_pre10_sync4_pay8_433mhz.h"
#else
    #include "configs/radio_config_fixed_crc_pre10_sync4_pay8.h"
#endif
#endif
```

with just the config definition you need e.g. for 433mhz fixed length, try
` #include "configs/radio_config_fixed_crc_pre10_sync4_pay8_433mhz.h"`

I presume your config file is created using SiLabs WDStool? If not, then use this tool, load a suitable XML config definition supplied in this repository, then change just the frequency and make a new config header.

GIPdA, 2018-03-23:
Hi,
You can read part details so the issue certainly comes from your config file.

You can try first with the default 433MHz config. Then to get to 434MHz, juste use channel 4... (channels are 250kHz apart).

Or, as @beinnlora said, use one of the 433MHz XML files with WDS and change your settings as needed (careful the lib may not support everything). If both modules have the same config, it should work. begin() returns false if the init failed (can be a debug check for custom configs).

regisaabdalla, 2018-03-24:
HI!
I received some data, only this:

_CRC Error
Invalid Preamble
Detected Preamble
Detected Sync
Read IT status_
 
Could you give some directions? 
Perhaps I need to work more on config file!

regisaabdalla, 2018-03-24:
I find the problem!
The word size was not 16, need to be 8!

But a only received "> ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮" form EzRadio.
It´s normal?!

GIPdA, 2018-03-24:
Are you using the default 433MHz config?

regisaabdalla, 2018-03-24:
I believe so!
I am use "si4455_revb1_bidir_fixed_crc_pre10_sync4_pay8_433mhz" file that comming with LIB.

GIPdA, 2018-03-24:
But did you use the default header (#include "configs/radio_config_fixed_crc_pre10_sync4_pay8_433mhz.h") and/or the #define ZETARF_FREQUENCY_433MHZ 1 to load the default 433MHz header config?
Or did you used the XML file to create your own config header? If so, I suggest starting as I suggested in my first comment (using the default config).

regisaabdalla, 2018-03-24:
I follow your suggestion and change the ZetaRF.CPP, as you describe in your first post.
I will double check my project!

regisaabdalla, 2018-03-24:
I double check all firmware and connections, but I just can't listen the message from EZRadio stick ... :-(

GIPdA, 2018-03-24:
What is your hardware setup? Can you try with two EZRadio sticks, or two Zeta? To be sure that the firmware is working ok.

regisaabdalla, 2018-03-24:
I have two EZRadio sticks, and they work perfectly.
But I can get communication with ZETA...

GIPdA, 2018-03-24:
Both have a SI4455 chip? Which revision? (Different revisions may need other settings. Print part info for both modules and compare.)
Do you have two Zeta modules to try?
Both modules have a similar antenna? (same polarisation, etc)

I can't really help more than giving you some "clues" to look for, I don't own any 433MHz module. But if the firmware and chips are the same for both EZRadio stick and the Zeta, it should work.
If the revisions are different, you need to cross-check with WDS to get the same config on both modules.

regisaabdalla, 2018-03-24:
you are a genius!!!  "...Which revision? " ... that's it!!
the zeta is B1 and EZradio is C2.
It's alive, doctor!!  It's alive......
**THANK YOU!!!**

regisaabdalla, 2018-03-24:
But... when I send something from ZETA to EZRadio The arduino freezes .... :-(
Any suggestions?

GIPdA, 2018-03-24:
Did you changed something on the config then?
What code are you using? The provided example? Which "side" freeze ?

regisaabdalla, 2018-03-24:
Yes I change the config. I use WDS to generate a config file compatible with Demo APP "Bidirectional packet". Now I can listen all communication between two EZRadios. But when I send something the code freezes with no response on "_zeta.sendPacket((const uint8_t*)data);  // Use channel set with begin()_" from your code.

I do not change your code, only the channel to 0. : " _zeta.begin(0, ZETARF_PACKET_LENGTH);_"

GIPdA, 2018-03-24:
You're not very clear.
Listen & send between two EZRadio works, right?
EZRadio (send) to Zeta (receive) works?
Zeta (send) to EZRadio (receive), Zeta freeze?

regisaabdalla, 2018-03-24:
Sorry ....
_Listen & send between two EZRadio works, right?_  Yes,no issues with this.
_EZRadio (send) to Zeta (receive) works?_ Yes, I can read the transmission no matter wich EZRadios transmit.
_Zeta (send) to EZRadio (receive), Zeta freeze?_ YES, this is the problem!

GIPdA, 2018-03-24:
Ok, thanks.
With the Zeta, you're using the default config header provided with the repo, right? Your config for the EZRadio will not work with the Zeta (different chip revision). You're not very clear either with what you did with the config ;)

You need to keep the default Zeta config, and create a new config with WDS for your EZRadio that match the settings of the default 433MHz Zeta XML: load the provided XML with the B2 revision, then load the same XML but with a C2 revision, then compare both configs and make changes on the C2 revision to match settings.
You can also check what changed in the C2 revision.

regisaabdalla, 2018-03-24:
OK! I will make this verification. 
Thanks!

GIPdA, 2018-03-24:
Ok, let me know how it goes. It will be useful for others using later revisions too.

regisaabdalla, 2018-03-24:
I verify all code without note any difference. 
I decided come back to original version.
The result was the same.... this is the results:

Starting Zeta TxRx...
Chip rev: 17
Part    : 21828
PBuild  : 0
ID      : 3328
Init done.

Type AAA and hit "enter"

Sending >aaa
<

.... NOTHING after this point

May be the ZETA is defective?

regisaabdalla, 2018-03-24:
New result....

with: "wwwwwwwwwwwwwwwwwwwww"
I get this:
**Sending >wwwwwwwwwwwwwww
<
msg transmitted
Sending >
<**
then freezes again...

GIPdA, 2018-03-24:
When the sending worked, did you received anything on the receiver?

About the freeze, I would say wrong config... you can enable debug by uncommenting the line "#define ZETARF_DEBUG_VERBOSE_ON" in ZetaRF.h, and maybe add debug in the sendPacket() method to check where it stalls exactly. That would help isolate the issue.
There are checks to avoid stalling but it seems like I missed one at least.

regisaabdalla, 2018-03-24:
OK! I will proceed with tests...

regisaabdalla, 2018-03-24:
Before freezes I received:

<Read IT status
Invalid Preamble

GIPdA, 2018-03-25:
Add some debug in sendPacket method, something like:

`void ZetaRF::sendPacket(uint8_t channel, const uint8_t *data, uint8_t length)
{
    if (!data || length == 0) return;
    // Read ITs, clear pending ones
    readInterruptStatus(0, 0, 0);

    if (m_systemError) return;
    
    Serial.println("check state");
    // Wait when not ready
    DeviceState s = deviceState();
    uint16_t counter = 0xF000;  // Avoid stalling
    do {
        --counter;
        s = deviceState();
    } while ((s == Tx || s == TxTune) && 
             counter > 0);

    Serial.print("state: ");
    Serial.print(s);
    Serial.print(" / ");
    Serial.println("counter");

    if (counter == 0)
        return;

    Serial.println("write tx");
    // Fill the TX fifo with data
    writeTxFifo(data, length);

    Serial.println("start tx");
    // Start sending packet on channel, return to RX after transmit
    startTx(channel, 0x80, length);
}`

regisaabdalla, 2018-03-25:
I will....and post results as soon as possible.
TkS!

regisaabdalla, 2018-03-30:
HI! Its working!!!
You will not  believe .... I'm so stupid!!

I was using 5V to Arduino and 3.2V for ZETA, because we can't program arduino with 3,2V.
A decide try a change, and apply 3.2V on Arduino...BANG!!!  WORKS like a charm.
Probably some problem with logic levels conversion.

Thank you for the support, and congrats for the code!!!

Best regards,
Regis

GIPdA, 2018-03-31:
Hi, glad you solved the problem! :)
I'll add a warning in the readme about that.

Thanks,
Best regards,
Benjamin

closed 2018-03-31 by GIPdA

regisaabdalla, 2018-04-06:
Hi, Benjamin !
I moving forward to a next phase in my project and decide to use a more powerful microcontroller.
I will use a PIC24. Now I will try to port your library to PIC - 'C'.
Do you have any tips and recomendations or "perhaps" an "PIC version" of your code?
Could you help me? Please!

Thank you in advance!
Best regards,
Regis

GIPdA, 2018-04-06:
Hi Regis,
I don't have any advice to give you, and I don't have a C version of the lib. I almost exclusively use C++ and I avoid PICs so I can't help you. You can maybe use the codebase of the [Wireless Development Suite](http://www.silabs.com/products/wireless/EZRadio/Pages/Si4455.aspx), this lib is based on that. Or adapt this lib, whichever you prefer. You'll have to adapt SPI, GPIOs, delays, and maybe some C++ stuff.
Good luck!

regisaabdalla, 2018-07-01:
Hi, Benjamin ! I hope this message finds you well and happy!
I successful finished that project . **THANK YOU!**

And ... start a new one! 
Do you have any ideas about why an arduino nano + zeta clone (revC1) freezes after 8 minutes of normal operation?

Tks!

GIPdA, 2018-07-02:
Hi Regis ! Nice, great to hear :) I guess you successfully ported the lib to your PIC?

The freeze could come from a lot of things... memory leaks might be one. If it's linked to the Zeta, it may be due to the module resetting or getting non-responsive and failing to communicate, usually caused by power supply issues (the lib times out after a too many tries and sets an error you can get via systemError(). Try calling begin() again in that case).
In any case, try to isolate the problem by removing code gradually ;)

regisaabdalla, 2018-07-02:
> ... I guess you successfully ported the lib to your PIC?

Actualy... not! I am add another MCU (328P) to the project, than work like a charm.
I will try call "begin" when freezes using the watch dog and make a deep code study at same. 
I will keep you informed about my progress.

Thank you!

beinnlora, 2018-07-02:
@regisaabdalla  Arduino Nanos get their  3.3v via the USB chip - which for cheap clones is a CH340 chip, which are rated to supply just 12mA @ 3.3v. (see https://cdn.sparkfun.com/datasheets/Dev/Arduino/Other/CH340DS1.PDF)
The Zeta module uses 10mA at RX standby, and between 18 and 30mA during TX.
Depending on your duty cycle, you could easily be overstressing the 3.3V supply from the Arduino, which over time cuts out fromoverheating.
You can't really reliably power a Zeta from the 3.3v rail from a Nano. Use a separate 3.3v regulator such as LM1117-3.3

regisaabdalla, 2018-07-03:
Very good point!!! It Will make the modificativos André repeat the teste.
Thank You!!