GIPdA/ZetaRF #4: STM32F103RC support

okwestern, 2018-04-20:
I'm seeking this library for use with Arduino and the STM32F103RC.

GIPdA, 2018-04-20:
Hi,
Did you tried it? It should work no problem, as long as Arduino SPI & GPIO are supported on the core you're using for the STM32.

okwestern, 2018-04-20:
Arduino SPI and GPIO are supported. With the core I'm using, a SPI flash chip is responding to simple SPI communication, but the SPIFlash library that comes with Arduino does not work. Same with a ILI9341 display library. A special library is under development, so is an SPIFlash library.
I'm not sure what is wrong with my SI4455 setup, but when I'm reading chip info, I'm getting this back.
Info for 433MHz chip:
Chip rev: 255
Part    : 65535
PBuild  : 255
ID      : 65535
Customer: 255
Rom ID  : 0
Bond    : 0
Rev Ext   : 255
Rev Branch: 255
Rev Int   : 255
Patch     : 65535
Func      : 255
SVN Flags : 255
SVN Rev   : 16777215

GIPdA, 2018-04-20:
Probably a SPI issue then, it seems like you get only ones. zeta.begin() should also return false.
Best way to debug would be with a logic analyzer I guess.
Which core are you using?

okwestern, 2018-04-20:
I'm using STM32F1xx by stm32duino (v. 2018.3.18).

okwestern, 2018-04-20:
Another thing: stm32duino core does not define SPI transfer function with two arguments (used in line 1508 in ZetaRF.cpp: SPI.transfer(data, count)). Arduino does only have transfer(void *buf, size_t count). When porting Arduinos transfer function, stm32duino has not defined the adresses for SPI status/data register, interrupt flag (SPSR, SPDR, SPIF), etc.

GIPdA, 2018-04-21:
The SPI lib of the stm32duino core has similar methods:
`void transfer(void *_buf, size_t _count, SPITransferMode _mode = SPI_LAST)`
`byte transfer(uint8_t _data, SPITransferMode _mode = SPI_LAST)`
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SPI/src/SPI.h

Do you have a compiler error on that line?

I don't understand what you mean in your last sentence. The lib don't manipulate raw SPI registers.
I guess their lib should be compatible with the "original" Arduino one without modifications, otherwise it wouldn't be very smart. ZetaRF doesn't use anything special for the SPI, only basic send/receive.
Try reduce the SPI speed maybe?

GIPdA, 2018-05-02:
Ok, despite the names being a bit different, this lib also supports the same transfers: read(uint8 *buffer, uint32 length), write(const uint8 *buffer, uint32 length) and transfer(uint8) (lines 211, 230, 240).

You would do something like: (and similarly for `spiReadWriteByte` and `spiReadData`)
`void ZetaRF::spiWriteData(const uint8_t* data, uint8_t count) const
{
    SPI.write(data, count);
}`

Note that the method `ZetaRF::spiReadWriteData` isn't used anywhere and can be removed.

pauliusra, 2019-03-03:
Hello, I'm using 2 zeta 433D-rf,  1 arduino uno and 1 arduino nano as microcontrollers. I made sure to connect everything as mentioned in the table below this library, but the problem is that I can't receive messages. I'm using examples from this library, sending messages via serial monitor, both rf modules have 433 mHz antennas. It seems that transmitting process works fine. Rf modules powered by 3.3 V, also I added line #define ZETARF_FREQUENCY_433MHZ 1
Am I missing something?

GIPdA, 2019-03-03:
Hello @pauliusra, please open a new issue (button on top of the page).

GIPdA, 2021-02-01:
Closing this because I will not add support for the linked library and the official ST Arduino Core works without issues with the STM32F103.
Feel free to reopen if needed.

closed 2021-02-01 by GIPdA