GIPdA/ZetaRF #5: Problems with receiving zeta433d rf

pauliusra, 2019-03-03:
Hello, I'm using 2 zeta 433D-rf, 1 arduino uno and 1 arduino nano as microcontrollers. I made sure to connect everything as mentioned in the table below this library, but the problem is that I can't receive messages. I'm using examples from this library, sending messages via serial monitor, both rf modules have 433 mHz antennas. It seems that transmitting process works fine. Rf modules powered by 3.3 V, also I added line #define ZETARF_FREQUENCY_433MHZ 1
Am I missing something?

GIPdA, 2019-03-03:
Hello,
I never tested the 433MHz version but it was used successfully (previous issue/pull request).
Two things you can do:
1/ Check that zetarf.begin(..) returns true (the example would need an update). If not, probably a wrong wiring.
2/ Double-check your wirings, particularly nIRQ and nSEL, and that MISO/MOSI are not inverted.

Do you get the "msg transmitted" message?
Make also sure the boards use the same channel (try other channels if needed).

Also, check that `#define ZETARF_FREQUENCY_433MHZ 1` is **BEFORE** the `#include <ZetaRF.h>`. (it's also a thing to fix)

pauliusra, 2019-03-03:
Thanks for your swift reply.
1)ZetaRF.begin(..) returns true .
2)Wiring seems ok.
Yes I get message "msg transmitted".
I made sure to use same channel and tried another channels.
define ZETARF_FREQUENCY_433MHZ is before #include <ZetaRF.h>
Still can't receive a message.

GIPdA, 2019-03-03:
From past issues, other things to troubleshoot:
- 5V Arduino with 3V3 Zeta can cause problems, try with a 3V3 Arduino or use level converters
- Maybe an issue with the chip revision of the Zeta. If you uncomment the code to print chip info in the example, what do you get? (past the text here)
- 3V3 power supply may be too weak with some cheap Arduino boards (RX needs at least 10mA and TX about 20mA)

And just to confirm:
- you're using the latest version of the lib
- begin() returns true on both sides
- startListening() is called on RX side
- checkReceived() is called on RX side
- antennas are well connected

Check if any error occurred with systemError() on both sides.
Add `#define ZETARF_FREQUENCY_433MHZ` in ZetaRF.cpp, on top, to be sure that the 433MHz config is loaded.

You can try to add serial debug in checkReceived(), the radio link is maybe unreliable and the messages gets corrupted and dropped (unlikely, but... better check).

pauliusra, 2019-03-04:
Thank you for your help. The problem were antennas. I tried to check and fix almost everything what is possible, but not antennas, because  I bought them and soldered perfectly just from the beginning, they looked legit. Now I changed them to 17.3 cm wires and transmitting, receiving works perfectly. Conclusion- never trust Chinese sellers.
Thank you for creating library and helping people.

GIPdA, 2019-03-04:
Good to read you solved your issue. What type of antennas were used originally?
868MHz modules seems a bit more "reliable", I can get communications to work (on small range, some meters) even with bad or no antennas.

If you ever get to test your modules on long range or crowded environments, please let me know any results you get, I'm interested on what can be achieved with these and their limits.

Also, the radio config is currently a compromise of range and speed. I want to add in the future other configs and further improve range with lower bitrates and other stuff. If you (or anyone ;) ) ever want more range or maybe more reliability, feel free to open a new issue and I will try to give you other configs to test ;)

pauliusra, 2019-03-04:
Firstly I tried to use these antennas [https://www.aliexpress.com/item/10PCS-Spring-Antenna-RF-Receiver-Transmitter-Module-433-MHZ-For-Smart-Home-Light-Wireless-Remote-Control/32876087747.html?spm=a2g0s.9042311.0.0.27424c4ddTaYWo](url) . I thought they would work great for testing even though they are cheap. 98% positive reviews looked safe.
But finally I used  17.3 cm thick pieces of wire. I used these "antennas" before with great success.

433mHZ rf modules didn't work with incorrect antennas, however I'm not sure if they work without antennas. 

I will try to test these modules in the city area with houses and walls and probably these antenna wires I have now. I just wanted to have reliable rf module in a short range that's why I didn't care much about antennas and the maximum range. I will send you some info when I will get it.

Currently I'm not really interested in long range RFs, but I might be in the future. If I will need it I will know that modules and library to use :)
What range have you managed to reach? In the datasheet they wrote about >2 km.

closed 2019-03-05 by pauliusra