GIPdA/ZetaRF #6: Questions about other possibilities of zeta library

pauliusra, 2019-03-05:
Hello again. 
I'm new at github so I'm not sure if this is a good place for the following questions.
1) I tested zeta 433 mHz modules with this library and I got this data [https://prnt.sc/mtn3nu](url) My code worked for 0.8 s, zeta.begin 4.5 sec, 0.3 sec for sending a message. I want to use these modules as a switch so 4.5 sec becomes a problem. Is it possible to reduce that time? If yes, how much? Range doesn't matter that much.
2) Is it possible to use Si4455 without external mcu? For example it would be wonderful by raising GPIO1/2 to high and making it to send preinstalled messages.

GIPdA, 2019-03-05:
Hello
It is the right place.

1/ That's strange, it shouldn't take so long to initialize. Can you add debug in ZetaFR::begin() to know what part takes so long?

2/ No, not supported by the chip. It can't even be used without loading a config setup.

GIPdA, 2019-03-05:
Actually, please open a new issue for your 1., that way it will be easier to find if somebody ever has the same problem. Because it is indeed an issue to fix.

beinnlora, 2019-03-05:
Please do open a new issue for the slow time to first message. I had a similar requirement where I needed to transmit as soon as possible after powerup (battery powered setup) so have some code hints for getting around 250ms (I lose the exact timing) from powerup to sending a short message at 433MHz and 4800 baud

closed 2019-03-07 by GIPdA