GIPdA/ZetaRF #7: ZetaRF library working slow

pauliusra, 2019-03-06:
Yesterday I had a problem. Zeta.begin was working for about 5 sec on arduino mega. Today I tried with the same code, everything was the same but now zeta.begin worked for just 1.4 sec. That was strange...
After testing with arduino mega I tried uno and nano and I got much better results. Arduino nano from the powering to message sent 0.9 sec, arduino uno 0.9 sec, arduino mega 2.4 sec. Arduino nano and uno were Chinese versions while mega was original. After this test I tried using Chinese arduino mega, but with no success. It was keep looping void setup () even with 2 sec delays added. 
Then I tried using baud 4800 as suggested by beinnlora (before I used baud 115200 as a default in this library). And I was surprised to get just 0.16s from power up to message sent (I was using arduino nano). So its 5.6 times faster than with baud 115200 and 15 times faster than arduino mega.
Test [http://prntscr.com/mtwnqq](url)
So I guess arduino mega should be avoided while testing and for better speed baud 4800 is a good choice. I will try to improve speed further, however why did my arduino mega worked 3 times slower yesterday than today with the same connections, same computer, same code?

GIPdA, 2019-03-07:
Just tested "boot" time (just Zeta.begin()) with a Chinese Arduino Uno and a 868MHz Zeta: **143.24ms** (repeatable)
Serial at 115200. Serial baudrate doesn't change anything, as expected (tested with 9600 and 4800). If anything, the lower the baudrate the longer it will take overall to boot... (zeta.begin() will not change, it is not linked with the serial)

> however why did my arduino mega worked 3 times slower yesterday than today with the same connections, same computer, same code?
How do you trust your terminal to print accurate timestamps? Something changed or it should have been the same. Mega or not, if the clock speed is the same it should be quite the same too.

I don't have a 433MHz module, but you can use my test code to compare (from the example, nothing else was modified):
```
  unsigned long t = micros();
  // Initialize Zeta module, specifing channel and packet size
  zeta.begin(2, ZETARF_PACKET_LENGTH);
  unsigned long d = micros() - t;
  Serial.print("Begin time: "); Serial.print(d/1000.); Serial.println(" ms");
```

beinnlora, 2019-03-07:
how are you providing 3.3v power to the Zeta?

pauliusra, 2019-03-07:
Things are getting more and more mysterious. I'm sure that I haven't changed the code and baud rates doesn't matter now. I was saving my time results and they clearly showed that baud rate did matter, but it was clear only on arduino mega, uno and nano had no issues. Today I tried testing working time on mega and at first it was keep looping void setup. So I decided to try arduino nano which showed result of 144 ms. After that I was trying to test again on mega, I put delay after serial begin, but it didn't help. So I removed delays. After that I tried changing baud rates and uploading the same code and restarting mega a few times and it started to work without looping the setup. Result 4453.64 ms, I don't know why mega can't work correctly, It's acting in a strange manner, it is possible that I got different results while changing baud rates just because of these mysterious bugs. 
Results: original arduino mega [http://prntscr.com/mupwct](url) 
Chinese arduino mega [http://prntscr.com/mupwtu](url)  
To be honest I have perfectly working code (with uno and nano) so I can leave everything as it is and not investigate any further. But if you want me to carry out this testing further I can do that.

bennlora I used 3.3 V pin from arduino to power Zeta.

GIPdA, 2019-03-07:
The mega may not be able to supply enough current. Do you have anything other than the Zeta module on the 3V3 rail? According to the doc, the 3V3 output is limited to 50mA. A Zeta module might need up to 30mA or so (10mA RX, 18mA TX).

Try to power the zeta with an external regulator (another (working) arduino maybe)? (don't forget to join grounds)

I'll test tomorrow if I can get my hands on a mega.

GIPdA, 2019-03-07:
(
@pauliusra Just a quick off-topic note, when adding an URL, GitHub adds this: [] (url), you need to replace the text "url" with the link, and put the name of the link (it can be the link itself) between the brackets [ ]. (I fixed the links in your previous posts, you can check to see (edit button).)

And a better alternative would be to attach them directly to your post (drag&drop in the editor), to avoid dead links in the future. Or if it's just text, past it in a code block.
Thanks! :)
)

pauliusra, 2019-03-08:
Sorry for your wasted time, because of my stupidity. The problem why it didn't work is in the poor connection. I forgot that mega have different MOSI, MISO and SCLK pins. After connecting everything properly everything works fine...

GIPdA, 2019-03-13:
Glad you found the issue at least ;) So you used the wrong pins on the mega, that's all? (just to be sure, for future ref)
And in your previous tests, the initialisation, despite being very long, succeeded or not? I guess no, but it sounded like it did...

pauliusra, 2019-03-13:
Yes, the problem was in the wrong pins and that's all.
I'm not sure about initiliasation, because I was just interested at a sending time at that moment, but probably it wasn't successful. 
P.S. I just tried using atmega 328 as a mcu for tranceiver. It's working as expected, everything works perfectly.

closed 2019-03-13 by GIPdA