GIPdA/ZetaRF #9: Issues with Teensy/433Mhz - begin(...) returning false

hallgchris, 2019-12-17:
Hello! I have been trying to run your library on a Teensy LC and 3.6 with the 433 MHz version of the Zeta. Unfortunately I've been unable to get anything to send yet - although I strongly suspect user error.

Here's serial output I get from running the example code in this repository (with `#define ZETARF_FREQUENCY_433MHZ 1` added before the ZetaRF include). I've also defined `ZETARF_DEBUG_VERBOSE_ON`.

```
21:27:42.221 -> Starting Zeta TxRx...
21:27:42.320 -> Read IT status
21:27:42.320 -> Invalid Preamble
21:27:42.320 -> CMD Error
21:27:42.320 -> Read IT status
21:27:42.320 -> Invalid Preamble
21:27:42.320 -> Read IT status
21:27:42.320 -> Invalid Preamble
21:27:42.320 -> FIFO Under/Overflow Error
21:27:42.486 -> Read IT status
21:27:42.486 -> FIFO Almost Empty
21:27:42.486 -> Invalid Preamble
21:27:42.486 -> FIFO Under/Overflow Error
21:27:42.486 -> CMD Error
21:27:42.486 -> Read IT status
21:27:42.486 -> FIFO Almost Empty
21:27:42.486 -> Invalid Preamble
21:27:42.486 -> Read IT status
21:27:42.486 -> Invalid Preamble
21:27:42.486 -> FIFO Under/Overflow Error
< Cropped output, continues like this for a while >
21:27:43.778 -> Read IT status
21:27:43.778 -> FIFO Almost Empty
21:27:43.778 -> Invalid Preamble
21:27:43.778 -> FIFO Under/Overflow Error
21:27:43.778 -> CMD Error
21:27:43.778 -> Read IT status
21:27:43.778 -> FIFO Almost Empty
21:27:43.778 -> Invalid Preamble
21:27:43.778 -> Read IT status
21:27:43.778 -> Invalid Preamble
21:27:43.778 -> FIFO Under/Overflow Error
21:27:43.778 -> ----------
21:27:43.778 -> Chip rev: 34
21:27:43.778 -> Part    : 85
21:27:43.778 -> PBuild  : 134
21:27:43.778 -> ID      : 1536
21:27:43.778 -> Customer: 0
21:27:43.778 -> Rom ID  : 0
21:27:43.778 -> Bond    : 0
21:27:43.778 -> 
21:27:43.778 -> Rev Ext   : 6
21:27:43.778 -> Rev Branch: 0
21:27:43.778 -> Rev Int   : 2
21:27:43.778 -> Patch     : 256
21:27:43.778 -> Func      : 0
21:27:43.778 -> SVN Flags : 69
21:27:43.778 -> SVN Rev   : 12285568
21:27:43.778 -> ----------
21:27:43.778 -> Read IT status
21:27:43.778 -> FIFO Almost Empty
21:27:43.778 -> Invalid Preamble
21:27:43.778 -> FIFO Under/Overflow Error
21:27:43.778 -> Init done.
```

`zeta.systemError()` returns `false`, but `zeta.begin(...)` is also returning `false`.

Any ideas what might be causing this? Thanks for your help :)
Chris

GIPdA, 2019-12-17:
Hello,
Thank you :)

I would guess a bad wiring? Lots of errors in the comm leads me to suspect something like that. Check that all is wired correctly and test the leads with a multimeter. Usual "Arduino/chineesium" bundles of wires can have some bad ones that are cut somewhere internally.
`begin()` returning false would mean that the module initialisation/setup failed (probably due to a wiring error).
Part infos doesn't look wrong, but it's hard to say.

How did you wired the Zeta to your Teensy?
The 433MHz module is known to work, so quite certainly user error, indeed ;) And I'm using a 868MHz one with a Teensy 3.6, so that should work too.

hallgchris, 2019-12-18:
Hey, just want to keep you updated, although it'll be a couple days before I can sit down and take proper look at what's going on. Thanks for your help by the way! Although I'm sure I haven't made any changes whatsoever since yesterday, I'm now getting a lot fewer errors, and I'm actually able to send messages between the teensys!

I'll try to work out what's going on myself when I have time, but for your interest, the output now looks like this (on both teensys):

```
21:46:04.435 -> Starting Zeta TxRx...
21:46:04.799 -> System Error!
21:46:04.965 -> Read IT status
21:46:04.965 -> Read IT status
21:46:04.965 -> begin() returned true
21:46:04.965 -> ----------
21:46:04.965 -> Chip rev: 34
21:46:04.965 -> Part    : 85
21:46:04.965 -> PBuild  : 134
21:46:04.965 -> ID      : 1536
21:46:04.965 -> Customer: 0
21:46:04.965 -> Rom ID  : 0
21:46:04.965 -> Bond    : 0
21:46:04.965 -> 
21:46:04.965 -> Rev Ext   : 6
21:46:04.965 -> Rev Branch: 0
21:46:04.965 -> Rev Int   : 2
21:46:04.965 -> Patch     : 256
21:46:04.965 -> Func      : 0
21:46:04.965 -> SVN Flags : 69
21:46:04.965 -> SVN Rev   : 12285568
21:46:04.965 -> ----------
21:46:04.965 -> Read IT status
21:46:04.965 -> Init done.
21:46:43.818 -> Sending >testing send
21:46:43.818 -> <
21:46:43.818 -> Read IT status
21:46:43.818 -> Invalid Preamble
21:46:44.050 -> msg transmitted
21:46:53.953 -> Invalid Preamble
21:46:53.953 -> Detected Preamble
21:46:53.953 -> Detected Sync
21:46:53.953 -> Invalid Preamble
21:46:53.953 -> Detected Preamble
21:46:53.953 -> Detected Sync
21:46:53.953 -> > testing recieve
21:46:53.953 -> 
```

Seems a bit bizarre to me, that both teensys can start working like that... Don't expect any help for now as I should look into it myself before wasting your time, but on the other hand let me know if anything's jumping out at you

beinnlora, 2019-12-18:
How are you powering your Zeta?
The Zeta module uses 10mA at RX standby, and between 18 and 30mA during TX at 3.3V.

The TeensyLC can only supply 5mA at 3.3V, the Teensy 3.6 10mA at 3.3V.
You'll need to use an supply of 3.3V external from Teensy to power the Zeta reliably

GIPdA, 2019-12-18:
Most probably some issues with your wires, you can try to move them a bit to see if it stops working?
System errors are bad but not critical, but means the comm is unstable or the module is malfunctioning.
Invalid preambles are normal.

GIPdA, 2019-12-18:
The Teensy 3.x boards can source 250mA on their 3V3 output. Teensy LC has 100mA, according to the pinout sheets: [https://www.pjrc.com/teensy/pinout.html](https://www.pjrc.com/teensy/pinout.html).
I/Os can only support 10mA.

beinnlora, 2019-12-18:
my bad: I had read the DIO current ratings here: https://www.pjrc.com/teensy/techspecs.html

hallgchris, 2019-12-18:
It's definitely sounding like dodgy wiring, I'll take a look at the wiring and let you know what I find. Interesting to hear that an invalid preamble is normal!

GIPdA, 2019-12-18:
The repeating pattern of the preamble can be detected for something that's not a valid frame, but also serves to synchronise with the incoming frame, which takes some time and errors before "locking in" :)
Also, as the current radio config is optimised for very low latency, the preamble is relatively short so the chance for errors in the preamble detection may be a bit higher than ideal. But for us at least it's been very reliable and not an issue :)

hallgchris, 2020-01-06:
Hey, sorry it's been so long. I've had a few gaps working on this project so my memory is a bit cloudy, but yeah seems like it was just some dodgy wiring. My best guess is it was a dodgy breadboard/jumper cables for one of the teensy/zeta pairs, and for the other, some bad soldering of the header pins on my part.

Thank you guys for your in-depth responses, sorry I've been so slow to get back to you all!

closed 2020-01-06 by hallgchris

GIPdA, 2020-01-06:
So worries, thank you for the update :)