GIPdA/ZetaRF #10: Sample app does not actually transmit

gRoberts84, 2020-02-19:
Hey there,

Apologies for what may be a totally obvious issue but I pulled in the sample app, connected my zeta 868 (https://www.rfsolutions.co.uk/radio-modules-c10/fm-zeta-transceiver-module-optimised-for-868mhz-p406) as per the pinout provided.

When running I can see msg transmitted in the monitor but when listening with an SDR I'm not able to see anything actually being transmitted.

Any suggestions?

GIPdA, 2020-02-19:
Hello!

The example should work, AFAIK. Are you able to test the RX with another ZETA 868 module?
Which SDR device are you using?

What are you looking for with your SDR? Apparent power spectrum? Zeta signal bandwidth is quite narrow, you're maybe missing it? Channels are 250kHz apart if I remember correctly, from 868.00MHz, signal bandwidth is quite smaller, maybe 50kHz?

You could also try to enable debug (in Zeta.cpp) and see if any errors shows up to assess if TX actually occurs.

Let me know what you find!

gRoberts84, 2020-02-19:
Hey there,

Thank you for getting back to me so quickly.

I'm currently using an RTL SDR with GQRX (it's able to do -/+ .500mhz I think) to visualise the transmitted data in a waterfall. 

I have a number of 868mhz devices that shows up perfectly so assumed it would be perfect to ensure my Zeta transceiver is able to transmit successfully.

Just in case, I also modified the sample to send every 1 second just in case.

I'll dig it all out and set it back up to see if I can provide more information.

Thanks again

GIPdA, 2020-02-19:
I would love to try, but I don't have any SDR receiver :(
Would it be possible that the bandwidth is so small that your receiver is not picking it up? Or more probably too fast, the Zeta is configured to have very low latency, with way less bytes over the air that other "normal" transmitter. Just a guess :P
Try to send way faster, like 50-100Hz? Or use zeta.checkTransmitted() with no delay for fastest throughput.

gRoberts84, 2020-02-20:
Hey there, sorry for the delay in getting back to you.

I suspect there is something a bit off as the `startup` method is being fired repeatedly and when uncommenting the code that outputs the chip details, I'm assuming I'm seeing erroneous data, as they:

```
----------
Starting Zeta TxRx...
----------
Chip rev: 255
Part    : 65535
PBuild  : 255
ID      : 65535
Customer: 255
Rom ID  : 255
Bond    : 255

Rev Ext   : 255
Rev Branch: 255
Rev Int   : 255
Patch     : 65535
Func      : 255
SVN Flags : 255
SVN Rev   : 4294967295
----------
Starting Zeta TxRx...
----------
Chip rev: 255
....
``` 

I've enabled the `ZETARF_DEBUG_VERBOSE_ON` flag, but that isn't outputting anything. Do I need to set this within the cpp file?

GIPdA, 2020-02-20:
No worries!
Indeed, the chip isn't responding (0xFFs everywhere, data line is kept high). begin() should fail and return false. Not sure if the example is checking that... (bad, I know!*)
You probably have something wired wrong or a bad cable if I have to guess.

Sorry the define is there to uncomment in the ZetaRF.h file, line 26. Not in the cpp as I wrongly said, and must not be in your sketch either.

*a v2 of the lib will be out shortly (it's in the v2 branch), but I still need to tweak some things and update/add examples to help identifying errors like that.

gRoberts84, 2020-02-20:
Thanks for getting back to me so quickly, response times like this on Github are rare!!

Does this look correct in terms of the wiring for an Arduino Uno R3:

Zeta     =>     Ardunio
2           =>     GND
3           =>     9
4           =>     3.3v
5           =>     8
9           =>     13
10         =>     11
11          =>     12
12          =>    10

Apologies for such a stupid question, but in terms of enabling the debug mode, do I need to uncomment the flag and rezip it before adding the library?

GIPdA, 2020-02-20:
You're welcome, you so happen to ask at just the right time ;)

Wiring pinout looks good, assuming you still have `ZetaRF zeta(10, 9, 8);  // Pins: SPI CS, Shutdown, IRQ` as in the example.

However, the Uno has 5V I/Os, and the Zeta is strictly 3V3, its I/Os are not specified to work at 5V. Although I've seen Zeta modules work at 5V (powered), it is very much not recommended and you could get different results (i.e. let the smoke out)!
If you can, try with a 3V3 I/O Arduino (like teensy or ESP), use a level shifter, or in least resort add a series resistor on each of the Arduino's output I/O (like 2k to 10k maybe) to limit current (would be MOSI, nSEL, SCK, SDN).

You can simply modify the file in your Arduino library folder if you unzipped the library there. If you used the Arduino IDE library manager as I suspect, I'm not sure where it puts the files, but you could modify, rezip and import again, yes. It should overwrite the files.

GIPdA, 2020-02-20:
Indeed the debug mode is quite the hell when something doesn't work properly... :P

gRoberts84, 2020-02-20:
> You're welcome, you so happen to ask at just the right time ;)

Good to know. Either way, glad you're here to help! It is much appreciated.

> Wiring pinout looks good, assuming you still have `ZetaRF zeta(10, 9, 8); // Pins: SPI CS, Shutdown, IRQ` as in the example.

That's the one. I made sure I followed the code pinouts based on your sample code.

> However, the Uno has 5V I/Os, and the Zeta is strictly 3V3, its I/Os are not specified to work at 5V. Although I've seen Zeta modules work at 5V (powered), it is very much not recommended and you could get different results (i.e. let the smoke out)!

Ah.. Now this is probably the issue. I haven't seen any "magic smoke" but I'd imagine given the size of it, it wouldn't give out much.

> If you can, try with a 3V3 I/O Arduino (like teensy or ESP), use a level shifter, or in least resort add a series resistor on each of the Arduino's output I/O (like 2k to 10k maybe) to limit current (would be MOSI, nSEL, SCK, SDN).

I'll try the resistor route and see if that will shed any light. Failing that, I think I will have to look at alternative radio units, i.e. cc1101, as they are cheaper and withstand the 5v without having to change anything.

> You can simply modify the file in your Arduino library folder if you unzipped the library there. If you used the Arduino IDE library manager as I suspect, I'm not sure where it puts the files, but you could modify, rezip and import again, yes. It should overwrite the files.

Gotcha, I found the folder with all the libraries I had imported via zip and was able to delete and add them back in with the modification. Now I know where they are I can modify them and upload accordingly.

Sods law, all of this is to try and hack my 868mhz thermostat. I have a load of other stuff I bought (Sonoff TH10) to control my boiler via an app, but the underlying issue is it would be next to the boiler much like my current thermostat which results in the heating going off sooner than it needs to etc. 

Fingers crossed I can get this working on something else. 

Thanks again!

GIPdA, 2020-02-20:
Glad to be able to help! :)

The Zeta modules are quite resilient on "voltage abuse". I had once some hooked up to 5V supplies for quite some time before I noticed it but they were working perfectly. But a 3V3 supply and 5V I/Os might cause more issues. Not impossible to find a similar issue on the closed ones actually, I feel like I remember something like that...

The module isn't cheap indeed. I recently found this one : https://www.tindie.com/products/dorji_com/868mhz-915mhz-wireless-rf-si4463-module-drf4463f/
The chip is not the same (same series, a bit better), package is a bit inconvenient (pain in the ass to solder wires to that), but I bought some and I will add support for them soon.

I never used the TI chips, but they have lower consumption, for the cost of lower RX sensitivity.

You're sure you can "spoof" your thermostat with a Zeta module? Or is it just a test? Seems unlikely to work out of the box to me, but I wish you good luck ^^ 

Cheers,

gRoberts84, 2020-02-20:
This is my naivety showing.

Despite being technically minded (developer and have a family history is radio and electronics etc) I stupidly assumed that I could use any transmitter that supports the bandwidth I need and simply replicate the bits that were being sent/received.

Reason I went down this route was simply trying to use what was already there. I have an 868mhz receiver with a relay that controls the boiler and the transmitter is a crappy analog dial that determines whether to send the on or off signal.

The sonoff th10 I have is great but the same issue exists where the thermometer is right next to the boiler causing it to shut off sooner than it should with it naturally being warmer around there.

Heh. It's all for fun. I can buy a nest/tado/hive and plug it in within two seconds and be up and running but that's not as fun.

GIPdA, 2020-02-20:
Not as easy unfortunately, but I feel you! Hey it might actually work, who knows!
Knowing what chip is used on the RX side it could maybe be possible. Maybe ;)

As I see it, you could modify the thermostat (add a board and drive the relay?), or even roll your own.
With the sonoff you can also link it up to a "home managing system" (I'm sure there is a better name for that :p) running on a Raspberry and have much more options. Add other temp sensors everywhere, and problem solved? ;)

closed 2020-07-18 by GIPdA