GIPdA/ZetaRF #11: Raspberry Pi support

hallgchris, 2020-07-10:
Hey GIPdA!

I've recently been working on porting your `v2` branch to run on a raspberry pi for my own use. I was wondering, is this something you'd be interested in having in your project?

You can find my work so far [here](https://github.com/hallgchris/ZetaRF/tree/rpi-hal) if you're interested. It seems to be working, although so far it's still a WIP. Wasn't sure whether to make a PR or to just gauge interest first as I'm new to open source and not really familiar with etiquette... But if you're interested, I'll tidy it up and make a PR :)

Keen on hearing your thoughts!
Chris

GIPdA, 2020-07-10:
Hi Chris!

I would be interested, for sure, thank you! I took a look at it, it seems fine, the Arduino-like API simplifies things :)

However, the v2 branch is also a WIP still, and although it seems to work fine (and fixes some issues the v1 has), I do not consider it stable yet and the API is subject to change. Like the xx_Gpio1AsClearToSend HAL class, I know that it doesn't work as is (because I'm currently working on it). Just so you know that I may require you to update or test things later on ;)

But PRs are welcome yes, please do :) About tidying up, you'll have to fix the indentations, seems like there are a mix of tabs and spaces in the RPI HAL header ;) Convention for me is 1 tab=4 spaces.

Side note, you may have noticed the si4463 additions, if you ever look for a cheaper (but harder to solder) alternative to the Zeta modules, Dorji sells their DRF4463F modules in 433 and 868MHz (seen on Tindie).

Benjamin

hallgchris, 2020-07-10:
That's great to hear! I'll work on tidying things up. FWI, the inconsistent spaces in the HAL header seems to have been copied across from the Arduino header. I am happy to update and test things in the future to the best of my capabilities!

The si4463 looks quite promising, I might need to try it out in the future! Looks like it's both higher power, more sensitive and higher bandwidth in addition to cheaper... Doesn't seem to be a major trade off (to my inexperienced eyes anyway)

I will send in a PR soon!

GIPdA, 2020-07-11:
Hum, yes, I checked my files, it's my fault actually, sorry x) I'll fix them too.

Indeed the Si4463 has better performances, it uses the EzRadioPro "API", which also has more features I might take advantage of later. At the moment the configs are probably not compatible with the Zeta modules (not tested yet), but I'll work on that. I'm pretty sure it's possible because I did it with Si4468 transceivers in the past.
I also made a breakout PCB to 2.54mm headers for it, I'll add the gerber files in the repo, maybe you'll find that useful too ;-)

I'm also testing some new configs for longer ranges (with lower data rates, so higher latency), is that something you could benefit from? Just curious, maybe you don't use them in difficult places like me :-P I used the Zeta modules (868MHz) in dense forest and got about 400 to 500m range, and I would like a bit more.

hallgchris, 2020-07-11:
I'm yet to experiment with different radio configs, but I'm pretty sure I'm actually going to need higher data rates than I've gotten so far, probably around 1KB/s. (Still need to test what I can get out of the default config). Hoping to achieve this at a range of 500m or so with line of sight (but of course, longer range is better if possible!). I had a go at testing this range (with the default config) a few months ago and was mostly successful, although I recall dropping lots of packets towards 500m. I suspect this could perhaps be improved with a better antenna, as I was just using a 173mm copper wire (at 433 MHz). I'd be curious to know what sort of antenna you're using to get the same range with much more obstruction!

I'm going to write a few tools in the next few days to test bandwidth and range to experiment with different radio configs and perhaps antennas. Excited to see if 1KB/s at 500m is possible!

GIPdA, 2020-07-11:
From my latency tests (using the latency example sketch), but mainly with 868MHz modules, the default configs with 32bit packets should allow for approx. 4kbits/s (so ~500 bytes/sec). I tried doubling the default data rate to 256k a few days ago, but it was very unreliable. Maybe with different RF settings 1KB/s could be possible, but it seems a bit out of reach ;) Using multiple modules on multiple channels could be an idea :P Or very directive antennas if possible.

I mainly use stripped coax cables or copper wires for my mobile "tags", and for the receiver a good SMA omni antenna tuned at 868MHz with a short bandwidth (+2 or 3dBi around 868 only). I tried RF Solution's antennas but they performed much poorly. Can't tell you exactly where I got it, can't remember...
That was what I used for my tests in forest. I did the same test with LoRa modules a few years back, and oh boy the range was bad x)

Good luck with your tests then, let me know your results :) I'll do some myself too, in the following weeks.

PR #12 (2020-07-18): Add Raspberry Pi support
Hey, sorry this took a little while! Thought I should have a go using the library before I made the PR to check it's actually usable.

I'm using the CMake build system, mainly because it's the only thing I really no how to use. Do you have any thoughts on this? Any suggestions on how I could do this better, considering this is primarily an Arduino IDE project?

If you have an RPi lying around, you *should* be able to compile it with `cmake .` then `make`.

Happy to make whatever changes/improvements you can think of, so let me know and I'll try my best :)

(Fixes #11)
closed 2020-07-19 by GIPdA