GIPdA/ZetaRF #17: VariableLength Code transmitting but not receiving

emanuelefalli, 2023-06-15:
Hi, I uploaded the example code for Simple RxTx on two transceivers for 915 MHz, one is connected to an Arduino nano and the other one to UNO using a logic level converter. Everything works properly with the fixed length example code, however when I try the variable length code, I am able to transmit from one transceiver but nothing is seen on the other end. 
I also tried to activate the debug mode on one end but I don't receive any additional information regarding the receiving part.

I put a serial print right after the zeta.available() function, but it never goes past it. I'm not able to understand why there is such a difference from the fixed data package to the variable one.

P.S. I am testing the two transceivers without antennas as of now, it's not meant to but I just wanted to get everything right before soldering an antenna.

[image]

GIPdA, 2023-06-15:
Hello,
What code are you using with the variable length config?

emanuelefalli, 2023-06-15:
I’m using the example code for simplerxtx with variable length

GIPdA, 2023-06-15:
You mean the example "SimpleRxTxVariableLength", not "SimpleRxTx" modified with a variable length config, right?

Did you make sure both Arduinos run the same example code, on the same channel?

emanuelefalli, 2023-06-16:
Hi, yes I meant SimpleRxTxVariableLength. I didn't have my laptop with me so couldn't check straight away. But yeah I reuploaded the code to both devices again and can't see any data package received on the other end. The channel is number 4, the one present by default.

[image]

GIPdA, 2023-06-16:
Ok, I found the issue. Inspecting the commit history, seems like I messed up a bit with the big update for the Si4463 and mixed up some code. I mainly use the Si4463 modules so I didn't caught that.

Before I commit and push a fix, please try this fix to confirm it's working:
In ezradio_si4455.hpp, line 312, replace the following method: (basically it's '_length+1_' instead of 0)
```
    void startVariableLengthTx(uint8_t channel, uint16_t length, RadioState stateWhenTxComplete)
    {
        startTx(channel, length+1, stateWhenTxComplete);
    }
```

GIPdA, 2023-06-16:
Definitely working on my side. I pushed corrections, please update your library with the current master branch and try again. If still not working, switch back to fixed-length with the SimpleRxTx example to confirm it's still working as before.

emanuelefalli, 2023-06-16:
Got it!! I updated the library and noticed a mistake in my previous code. I was using 
`ZetaRf915<ZetaRf::nSEL<10>, ZetaRf::SDN<4>, ZetaRf::nIRQ<2>> zeta;` 
to initialize the Arduino Uno, but I didn't notice that the initialization for variable length was different. Now it works perfectly. 

Thank you so much!

GIPdA, 2023-06-16:
Great! :)