GIPdA/ZetaRF #18: ZetaPLUS 868MHz - UART wiring and code

MtK50, 2023-12-06:
Hello,

I was wondering if the library supports the ZetaPLUS module using UART communications? I successfully made 2 modules work at 868MHz using SPI; however, when I send the data (using the SimpleRxTx example) from my Board1 (Elegoo UNO R3),

```
11:33:24.616 -> TX >values
11:33:24.616 -> <
11:33:24.654 -> Packet transmitted
```

I receive this on Board2 (Kuman UNO R3):
```
11:33:24.655 -> Packet received with RSSI: 216
11:33:24.655 -> RX >��������<
11:33:24.655 -> HEX >
11:33:24.655 -> FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF<
11:33:24.655 -> Error: Device Busy! Restarting...
```

When I switch their roles,

Board2:
```
11:36:58.871 -> TX >Hello 
11:36:58.871 -> <
11:36:58.871 -> Error: Device Busy! Restarting...
```

And nothing is received on Board1.

I tried using a Leonardo instead of the problematic board (same pins), I get the same issue using the TM4C123 board:

```
11:47:38.699 -> Starting Zeta TxRx...
11:47:38.929 -> Failed to initialize the radio module!
11:47:38.929 -> ZetaRf begin failed. Check wiring?
```

Second part of my question:

Does the library support UART communication, and do you have any code examples to make it work?

Thanks a lot because I've been struggling for too long using 868MHz RF modules such as Zeta or MRF89.

(I can give you links if wanted :)
Kylian

MtK50, 2023-12-07:
Hello again,

Thank you for providing the details about the library. To be honest, I'm not a big fan of AT commands (mostly because I don't know them at all), but I will take a look soon and try things out. First, I want to ensure that SPI not working at all before officially exploring other solutions

I somehow "fixed" the `Error: Device Busy! Restarting...` problem by testing it at home (instead of at work like previous times (too much RF ??)). However, I still have a problem using the Leonardo ([this might be the issue](https://forum.arduino.cc/t/difference-between-arduino-leonardo-arduino-uno-r3-coding/565418/3)), and I must use a Texas Instruments Board (TM4C123 [official link to the board](https://www.ti.com/tool/EK-TM4C123GXL)) because it has the perfect number of pins, multiple UARTs and is powerful enough.

When I compile, I get this error:
Fixed by adding `#include <type_traits>` in ZetaRf.hpp
[image]

Then my wiring is:
2. GND -> GND   | 12. nSEL -> PD_1 (CS)
3. SDN -> PE_3    | 11. SDO -> PD_2 (MISO)
4. Vcc -> 3.3V      | 10. SDI -> PD_3 (MOSI)
5. nIRQ -> PE_2   | 9. SCLK -> PD_0 (SCK)
[image]
[image]

The program compiles great, but the problem is located in the initialization of `beginWithPacketLengthOf()` -> `if (!Base::beginWithConfigurationDataArray())` -> `while (!m_radio.loadConfigurationArray(configArray) && (retryCount--))`

```
17:16:09.944 -> Starting Zeta TxRx...
17:16:09.944 -> Initializing...
17:16:09.944 -> EZConfig Check error
17:16:09.944 -> Initializing...
17:16:09.944 -> EZConfig Check error
17:16:09.944 -> Initializing...
17:16:09.944 -> EZConfig Check error
17:16:09.944 -> Failed to initialize the radio module!
17:16:09.944 -> ERROR: beginWithPacketLengthOf FALSE
17:16:09.944 -> ZetaRf begin failed. Check wiring?
```

Do you have any idea of where (and mostly what) the problem is? Is it a library problem or an issue with the board/compiler (which is Energia)?

Thanks again; I'm available if you need more documents.

Kylian

MtK50, 2023-12-08:
You welcome :)

I would say 868 MHz, revision 1 (Brown? Orange?) 
[image]

On the µc : "455A CAU0 042"

GIPdA, 2023-12-08:
Ok so it's not the plus version, but it seems like it's a revision C2, which may not work with the B1 config the lib uses, and could explain the EZConfig error.

I joined a new config file for you to test, replace the existing file and use the ZetaRf868 config like in the SimpleRxTx example. Hopefully it'll work?

[config868_fixedlength_crc_preamble10_sync4_payload8.cpp.zip](https://github.com/GIPdA/ZetaRF/files/13619147/config868_fixedlength_crc_preamble10_sync4_payload8.cpp.zip)

MtK50, 2023-12-29:
Hello again!

Sorry for not responding for this long.. I manage to simply give up using SPI and then go to UART using the ZetaPLUS 868MHz.

I'm writing for future people looking for ZetaPLUS UART communication code example, and to close the issue

Following this (https://www.rfsolutions.co.uk/downloads/652412b497cb1739DS-ZETAPLUS-8.pdf)
[652412b497cb1739DS-ZETAPLUS-8.pdf](https://github.com/GIPdA/ZetaRF/files/13792841/652412b497cb1739DS-ZETAPLUS-8.pdf)

Wiring TM4C123GXL:

2. GND -> GND | 
3. SDN -> GND | 
4. Vcc -> 3.3V | 
6. TX -> PB_0 (Rx 1) | 8. RX-> PB_1 (Tx 1)

I created those 2 codes examples which send and receive datas on `channel 2` using a `3 bytes payload`
(I'm scared that ASCII character will not display correcty when you copy the code, just Google a decimal/ASCII converter then copy/paste in)

Sender:
```
String command = "";
String response = "";
long previousMillis = 0;

void setup() {
  Serial.begin(19200);
  Serial1.begin(19200);
  Serial.println("Init complete!");
}

int speedValue = 0;

void loop() {

  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= 5000) {
    previousMillis = currentMillis;

    /*
   
    Value         Name             Description
    0-15          Channel          Frequency channel, set in 250KHz increments starting at (FREQ dependent):
                                    - At 433MHz: 0=433.92
                                    - At 868MHz: 0=869.50
                                    - At 915MHz: 0=915

      1-64        Packet length     Defines the data payload length in 8-bit bytes.

                  Data              Your data to be transmitted. ZETAPLUS will then return to the
                                    state is was in prior to ATS command
    
    Example: Sending a 13-byte packet on channel 2

    Command :              A   T   S   2   13    DATA
    Decimal byte value :   65  84  83  2   13    Your 13 bytes

*/
    // ATS : command to send
    // STX ascii : channel 2
    // ETX ascii : 3 bytes long
    // byte 1 : ASCII ('c') // Decimal ('99')
    // byte 2 : ASCII ('speedChar') // Decimal ('speed')
    // byte 3 : ASCII ('c') // Decimal ('99')

    String part1 = "ATSc";
    String part2 = "c";
    int randomValue = random(100);
    // float randomValue = 6.45; // works fine with float (it just make it an Integer)
    char randomChar = static_cast<char>(randomValue);
    Serial.println(randomChar); 
    // Serial.println(randomValue);
    // Serial.print("ATV");
    Serial1.print(part1 + randomChar + part2);
  }
  

  if (Serial.available()) {
    String command = Serial.readString();
    Serial.println("Send command: " + command);
    Serial1.print(command);
  }

  while (Serial1.available()) {
    String response = Serial1.readString();
    Serial.println("Receive AT Command: " + response);

    Serial.print("  ASCII: ");
    for (int i = 0; i < response.length(); i++) {
      Serial.print(response.charAt(i));
      Serial.print(" ");
    }
    Serial.println();

    Serial.print("Decimal: ");
    for (int i = 0; i < response.length(); i++) {
      Serial.print((int)response.charAt(i));
      Serial.print(" ");
    }
    Serial.println();

    speedValue = (int)response.charAt(5);
    Serial.println("Speed Value: " + String(speedValue));
  }
}
```

Receiver:

```
String command = "";
String response = "";

void setup() {
  Serial.begin(19200);
  Serial1.begin(19200);
  Serial.println("Init complete!");

/*
  Value    Name          Description
  0-15     Channel       Frequency channel, set in 250KHz increments starting at (FREQ dependent):
                          - At 433MHz: 0=433.92
                          - At 868MHz: 0=869.50
                          - At 915MHz: 0=915

  1-65     Packet Length Length of the data packet to be sent in 8-bit bytes.
  
  Example: Enter receiver mode on channel 2 with a packet length of 10 bytes.

  Command :              A   T   R   2   10
  Decimal byte value :   65  84  82  2   10
*/

  // ATR : command to set the reception
  // STX ascii : channel 2
  // ETX ascii : 3 bytes long
  Serial1.print("ATR");
}

int speedValue = 0;

void loop() {

  if (Serial.available()) {
    String command = Serial.readString();
    Serial.println("Send command: " + command);
    Serial1.print(command);
  }

  while (Serial1.available()) {
    String response = Serial1.readString();
    Serial.println("Receive AT Command: " + response);

    Serial.print("  ASCII: ");
    for (int i = 0; i < response.length(); i++) {
      Serial.print(response.charAt(i));
      Serial.print(" ");
    }
    Serial.println();

    Serial.print("Decimal: ");
    for (int i = 0; i < response.length(); i++) {
      Serial.print((int)response.charAt(i));
      Serial.print(" ");
    }
    Serial.println();

    speedValue = (int)response.charAt(5);
    Serial.println("Speed Value: " + String(speedValue));
  }
}
```

closed 2023-12-29 by MtK50